crust-firmware/crust #18: Write a test to check if Crust boots

agom94, 2018-01-18:


agom94, 2018-02-11:
Is this checking to see if crust boots on the hardware?

smaeul, 2018-02-12:
Yes. Unlike the unit tests that can run on the host, this is an integration test that has to run on the device. You can look at the `Makefile` for how `DEBUG` is handled. You'll want to add another variable like it, called something like `TEST`. Then in `common/main.c`, after initialization and before the main loop, you can call a function to run the tests, something like:
```
#if TEST
run_tests();
#endif
```
This function would be declared in `include/test.h` and implemented in `common/test.c`. The entire contents of `common/test.c` (after the includes) would also be gated by `#if TEST`. For this first test, all you need to do is have the function print a distinctive message, like the one given in the project documentation prepended by "test" (so it can be easily `grep`ed for), e.g. `test: Boot: PASS`.

The info below does *not* apply to this test, because we're only able to report anything if it succeeds, so the info below is for *future* tests only.

Further tests would need to hook into other parts of the system, e.g. the interrupt handler, the SCPI API code, etc. They would also be gated by the `#if TEST`. They would call a function defined in `include/test.h` that prints the appropriate success message and sets a flag (the flag being private to `common/test.c`. The main `run_tests()` function would set up a delayed work item to check all of these flag variables after some time, and print the failure message for those where the flag remains unset.

We would then need to also write a script that runs on the device in Linux to trigger the client-side operations (asking for CPU off/on, etc.). Then finally, we would need to write a Makefile rule or script that runs on you dev machine that ties it all together:
- Build the firmware with `TEST=1`
- Run the firmware on some device
- Run the client script on the same device

PR #74 (2018-03-06): Crust boot test
Closes #18
closed 2018-03-06 by smaeul