crust-firmware/crust #28: docs: Add CONTRIBUTING.md

drewwalters96, 2018-01-18:


smaeul, 2018-02-12:
## Commits

Splitting work into commits: unless the commit is a global change (e.g. formatting, copyright year, renaming something), changes should be split into commits for each major area of the code they touch. For example, a pull request for bringing up a new piece of hardware should be split into 2-4+ commits, or more if it adds clarity.
- The first commit should add the headers and source files for the generic driver class, if one is being added. (e.g. `drivers/<class>/<class>.c` and `include/drivers/<class>.h`)
- The second commit should add the driver and its header (e.g. `drivers/<class>/<driver>.c` and `include/drivers/<class>/<driver>.h`)
- The third set of commits (one for each platform) should add the devices that use this driver to `platform/<platform>/devices.c`. If this driver supplies an additional platform-specific header, that should go in these commits.
- Finally, the device can be used from common code.

The firmware should compile, run, and not regress the test suite after every commit, so fixes to a commit should be squashed into that commit before merging, not placed at the end of the PR.

## Formatting

All files should follow the same formatting rules. `uncrustify` is used where possible, otherwise the following general rules apply:
- Lines should be no longer than 79 characters + a newline.
- Use tabs for indentation and spaces for alignment (with one exception in assembly below).
- No trailing whitespace.
- All files must have a header at the top, using the appropriate comment style for that file type. Except for scripts, which have a shebang on the first line, the file header should begin and end with a line containing only the comment delimiter.

### C files

Functions in drivers should be topologically sorted (i.e. the probe function at the bottom). Other functions should generally be alphabetically sorted, with small `static` helper functions at the top. Functions and macros in headers should always be alphabetically sorted as much as possible for ease of finding them.

#### Naming in drivers

The driver should be named as generally as the hardware it supports. For example, if the driver supports the CCU on all sunxi devices, it should be named `sunxi-ccu`. If it supports the 64-bit counter _first introduced_ with the sun6i series, it should be named `sun6i-cnt64`. If it supports the SY8106A voltage regulator, it should be named `sy8106a`.

Note that hyphens are used in strings, filenames, and documentation. Underscores are used in variable names and header guards.

#### More header stuff

The header guard should be the same as the path you would need to put in your `#include` directive, but with the following changes:
- All upper case
- Hyphens, slashes, and periods replaced by underscores

### Assembly

Assembly files are preprocessed, so they should use the file extension `.S` (and this is the only extension the build system looks for). They should not have the same path as a C file (excluding the extension), as the build system cannot disambiguate the two.

Assembly should be formatted as follows:
- Local control-flow labels should be numbered, not named.
- Local data labels should be named descriptively.
- Global data and functions should use the `data` and `func`/`endfunc` macros from `macros.S`.
- Indent the instruction by one tab
- Indent the first operand with one tab
- Indent following operands with a single space
- Use `#` to start comments
- Align comments at 40 columns with tabs.

### Other files

smaeul, 2018-02-23:
### Tests
- Run unit tests on the host with `make check` (or `make test`).
- Verify formatting with `make check-format`.
- Run integration tests by building the firmware with `make TEST=1`

drewwalters96, 2018-02-23:
@smaeul Thank you for the additions

PR #94 (2018-03-05): project: Add community documents

## Purpose

The purpose of this pull request is to add community documents that establish a common procedure for contributing to the project.

Closes #28, Closes #30 

## Considerations for reviewers

None :)
closed 2018-03-05 by smaeul