crust-firmware/crust #44: Remove/restore power DRAM power on system suspend/resume

drewwalters96, 2018-01-30:
- [ ] If the system is to be put in an optional deep suspend state, the firmware will additionally remove power from DRAM on suspend, and restore power (and reinitialize the DRAM controller) on resume.

smaeul, 2018-01-30:
This requires putting the DRAM controller in self-refresh mode during suspend, as well as checksumming an area of RAM during resume (if the checksum doesn't match, switch to the normal boot flow (e.g. by resetting the system)).

smaeul, 2020-01-06:
Code for doing this for at least the H3 (which shares a memory controller with the A64/H5/H6) is in the Allwinner BSP; it may not be trivial to adapt it for crust.

smaeul, 2020-04-19:
Implemented in dda9582cb90b07348ec2a2e65c8295e73333f80f

closed 2020-04-19 by smaeul