crust-firmware/crust #56: Write unit tests for library code

smaeul, 2018-01-30:


drewwalters96, 2018-02-03:
@crust-firmware/crust Do we have a preferred testing framework for this?

smaeul, 2018-02-03:
For the C code, compiling `lib/` to a library (`libcrust.a` or whatever) and having a bunch of driver programs is simple enough. I don't know of any off the top of my head (I usually just write my tests as a bunch of asserts, ala https://github.com/smaeul/squish/tree/assn2.5/test), but you could probably find a small, simple C unit test library on the web.

Unit testing the assembly is going to require it to be run in an or1k emulator (e.g. qemu). For the math code, you'd need to come up with a bunch of random inputs. Feed them to both the code running in qemu (one version calling the functions directly, and a separate version using the C math operators), and a program on the host (using normal math operators). Then verify the output of all three programs is the same.

smaeul, 2018-02-11:
@agom94 I don't know why I didn't think of this earlier, but we also need tests for standalone library headers. (I missed them because they're in `include/`, not `lib/`.) This includes `bitmap.h`, `byteswap.h`, ~~`ctype.h`~~, `mmio.h`, and `util.h`. Please verify that the bit-operation macros and static inline functions work as described/expected.

PR #86 (2018-02-28): add tests for string library functions
Closes #56, Closes #82.
closed 2018-02-28 by agom94