crust-firmware/crust #169: Problem using compilers definition in Makefile

gatuno, 2019-12-30:
## Type of issue
Feature Request

#### Description

When using the current Makefile in crust, it may be hard to cross-compile for another platform, for example, imagine that you are on a amd64 host building an image for armv8a and using also the or1k compiler. The current definition at the main Makefile uses HOSTCC = cc, which could use the amd64 host's gcc, and not the armv8a that is needed.

My suggestion is to change the main Makefile to:
```
OR1K_COMPILE	?= or1k-linux-musl-
AR		 = $(OR1K_COMPILE)gcc-ar
CC		 = $(OR1K_COMPILE)gcc
CPP		 = $(OR1K_COMPILE)cpp
OBJCOPY		 = $(OR1K_COMPILE)objcopy

CROSS_COMPILE ?= 
HOSTAR		 = $(CROSS_COMPILE)ar
HOSTCC		 = $(CROSS_COMPILE)cc
```

Using the current Makefile would need to change the PATH variable to include first the cross-compilers (like the armv8a) and changing the Makefile would allow to specify when invoking the make.

smaeul, 2020-01-01:
In the past I've suggested using `make HOSTCC=aarch64-linux-musl-gcc tools`, but Kconfig is also built with `HOSTCC`, this now breaks if Kconfig isn't already built. I think what I want to do is use `HOSTCC` for the tools, and a new variable `BUILDCC` for Kconfig or other tools used during the build process. Then `HOSTCC` could default to an ARM/AArch64 toolchain.

smaeul, 2020-01-01:
This would require adding more rules to the top-level Makefile, and a new `buildprogs` Kbuild rule.

smaeul, 2020-04-19:
Fixed by 9e417af83b35f6882d3bbc2e1e183f9893c37158

closed 2020-04-19 by smaeul