crust-firmware/crust #171: Remove static inline wrappers from driver headers

smaeul, 2020-01-01:
## Type of issue

Enhancement

#### Description

The `<class>_driver` and `<class>_driver_ops` structures are implementation details of the drivers in that class, and should be moved from the public headers to a private header file in `drivers/<class>`. This involves converting the `static inline` wrappers to normal non-static functions, and converting the `<CLASS>_OPS` macro to a `static inline` function. See the the `clock` and `i2c` (https://github.com/crust-firmware/crust/commit/25e9355c2d93768aae195d86cdb8e2cdca4280be) classes for examples of how this is done.

Remaining classes where this is a TODO:
- `msgbox`
- `pmic`
- `regulator`
- `watchdog`

vincele, 2020-01-17:
Now the last one: pmic...

BTW, if you have other small tasks like this, please create issues for them, and I'll see what I can do.

smaeul, 2020-01-18:
Fixed as of #178

closed 2020-01-18 by smaeul