crust-firmware/crust #182: Makefile: please, remove HOST_COMPILE	?= aarch64-linux-musl-

xypron, 2020-11-02:
I try to compile crust on my ARM64 system. Obviously I will not need any cross compiler on the host and there is no reason to set HOST_COMPILE.

So why do you define HOST_COMPILE=aarch64-linux-musl- in this case?

smaeul, 2020-11-03:
I expect a majority of users to be cross-compiling from an x86 machine, where both the host tools and the firmware itself are cross-compiled. If you want to use the native compiler for building the tools, you can run `make HOST_COMPILE=`; or you can run `make scp` to skip building the host tools entirely.

I will consider adding autodetection logic to use an empty `HOST_COMPILE` if bare `gcc` appears to target `arm` or `aarch64`.

smaeul, 2021-12-16:
Now that loading and running Crust has been integrated fully into U-Boot and TF-A, the host tools are not very useful anymore (and I have not used them in a couple of years). I plan to disable building them by default. Would that change resolve this issue?

xypron, 2021-12-16:
My expectation is that I can build everything on the target device without patching any code. If that is given, I am fine.

smaeul, 2022-03-08:
`HOST_COMPILE` is no longer used by default.

closed 2022-03-08 by smaeul

Apteryks, 2023-07-07:
A related issue is that when building from a non-aarch64 architecture, the test objects fail to compile with "No cache cleaning implementation available for this architecture".  I had to patch out $(test-all) from the Makefile.

smaeul, 2023-07-07:
Which test objects? As of ac5d26580a322758d7dffaa8983946640a16a69e, `tools/test.c` will not be built by `make all`.

Apteryks, 2023-07-07:
I'm using the latest tagged release, 0.5, which is apparently older than that.

Apteryks, 2023-07-07:
Another thing I've noticed: if I *do* try to use `HOST_COMPILE`, e.g. by setting it to `aarch64-linux-gnu-` because I'd like to cross-compiled from my x86_64 system, the build fails like:
```
starting phase `defconfig'                                                                                on
                                                                                                          co
;;; (MAKE-FLAGS: ("CROSS_COMPILE=or1k-elf-" "V=1" "HOST_COMPILE=" "LEX=flex"))                            on
exec mkdir -p build/3rdparty/kconfig                                                                      co
exec mkdir -p build/include                                                                               ty
exec scripts/version.sh /tmp/guix-build-crust-pinebook-0.5-0.c308a50.drv-0/source build/include/version.h kc
exec cc -Ibuild/include -I./platform//include -I./arch//include -I./include/common -I./include/lib -D_XOPErtN_SOURCE=700 -I./3rdparty/kconfig    -Os -pipe -std=c11 -fdata-sections -ffunction-sections -fno-builtin -kcfno-common -fvar-tracking-assignments -g0 -Wall -Wextra -Wformat=2 -Wpedantic -Wshadow -Werror=implicit-function-declaration -Werror=implicit-int -Werror=pointer-arith -Werror=pointer-sign -Werror=strict-prototypes -Werror=undef -Werror=vla -Wno-missing-field-initializers -D_GNU_SOURCE -std=gnu11 -Wno-implicit-fallthrough -Wno-pedantic -Wno-sign-compare -Wno-unused-parameter -Wno-vla    -D_GNU_SOURCE -std=gnu11 -Wno-implicit-fallthrough -Wno-pedantic -Wno-sign-compare -Wno-unused-parameter -Wno-vla    -MMD -c -o build/3rdparty/kconfig/conf.o 3rdparty/kconfig/conf.c
/gnu/store/rib9g2ig1xf3kclyl076w28parmncg4k-bash-minimal-5.1.16/bin/sh: line 1: exec: cc: not found
make: *** [Makefile:174: build/3rdparty/kconfig/conf.o] Error 127
error: in phase 'defconfig': uncaught exception:
%exception #<&invoke-error program: "make" arguments: ("pinebook_defconfig" "CROSS_COMPILE=or1k-elf-" "V=1ar" "HOST_COMPILE=" "LEX=flex") exit-status: 2 term-signal: #f stop-signal: #f>                             mi
phase `defconfig' failed after 0.1 seconds
```
Apparently unless specified, HOSTCC and HOSTAR are overridden by BUILDCC and BUILDAR, which seems strange to me.

Apteryks, 2023-07-07:
This overridding appears to be caused by 
```
build_overrides  = $(foreach f,AR CC $2,$(eval $(obj)/$1: HOST$f = $$(BUILD$f)))
```
In the scripts/Makefile.kbuild file.

Apteryks, 2023-07-07:
so it seems when cross-compiling it's always necessary to use HOSTCC and HOSTAR, which is another reason why HOST_COMPILE is not useful (glad it got removed!)