crust-firmware/crust #187: Blocking Requests for Deep-Sleep

AsureDawn, 2021-02-09:
## Type of issue
Question or Feature Request

#### Description
I believe it would be useful, for ensuring updates complete without interruption and optionally to allow users greater flexibility in configuring power management, if system daemons could block Crust from going into deep sleep.

To that end, there should be some way to tell Crust not to enter sleep as it normally would, either via interrupt or configuration. Crust might also send out an interrupt to alert system daemons that it is about to put the device to sleep (and possibly whether it may be blocked or not)

This would allow, for example, systemd to check whether any units which conflict with "Sleep.target" are running, and either stop them gracefully or block the system from sleeping. It could also allow users to configure the device not to sleep when plugged into a power source, or an external monitor.

megous, 2021-02-09:
Look at Linux wakelocks API. That looks like that's what you're looking for.

Also this is only a problem if you've enabled the autosleep feature in the kernel and in /sys/power, otherwise the system will not go to sleep by itself, and only upon request of some system daemon. Typically systemd.

AsureDawn, 2021-02-09:
So crust already respects wakelocks (if supported/present)? Then, I guess I need to check my system daemon to see what support and configuration is available.

Thanks!

megous, 2021-02-09:
Crust doesn't do anything by itself. Suspend is always initiated by the kernel.

AsureDawn, 2021-02-09:
Got it! Thanks again. ^_^

closed 2021-02-09 by AsureDawn