crust-firmware/crust #188: Add Libretech / Libre Computer ALL-H3-CC defconfigs

rfinnie, 2021-02-27:
## Type of issue
Feature Request

#### Description

https://linux-sunxi.org/Libre_Computer_Board_ALL-H3-CC

I personally have the ALL-H3-CC H3 and ALL-H3-CC H5 (yes, the H5 variant has "H3" in the model name, sigh) and a merge request is forthcoming, but you may want to leave this open afterward as there's a few other variants.

smaeul, 2022-06-11:
Both of the mentioned boards were added in #189. A new issue could be opened for additional boards.

dsx724, 2022-06-13:
> ## Type of issue
>
> Feature Request
> #### Description
>
> https://linux-sunxi.org/Libre_Computer_Board_ALL-H3-CC
>
> I personally have the ALL-H3-CC H3 and ALL-H3-CC H5 (yes, the H5 variant has "H3" in the [...]

Sorry for the confusion. There's actually an ALL-H3-CC board with H3 or H5 that is basically the same as the ALL-H3-CC board but with gigabit ethernet and no CVBS.

smaeul, 2022-06-14:
I see an ALL-H5-CC (with H5) board in U-Boot, as `libretech_all_h5_cc_h5_defconfig`. Are you saying it should have the same config as ALL-H3-CC (with H5)? If so, I can make a copy of the config.

dsx724, 2022-06-14:
- ALL-H3-CC-H3, H3, CVBS, no GigE, libretech_all_h3_cc_h3_defconfig
- ALL-H3-CC-H5, H5, CVBS, no GigE, libretech_all_h3_cc_h5_defconfig
- ALL-H5-CC-H5, H5, no CVBS, GigE, libretech_all_h5_cc_h5_defconfig