crust-firmware/crust #193: NanoPi A64 support

sergey-suloev, 2021-09-30:
Should NanoPi A64 be supported by a separate config or defconfig should be used ?

smaeul, 2021-09-30:
From [the NanoPi A64 schematic](http://wiki.friendlyarm.com/wiki/images/c/c8/NanoPi-A64-1610-Schematic.pdf), the board looks to have the usual AXP803 PMIC and regulator setup, and it also has an infrared receiver. So this would be a good config to start with:
```
CONFIG_CIR=y
CONFIG_MFD_AXP20X=y
```
(which happens to be the same as `orangepi_win_defconfig`).

Let me know if this works for you, and I will add it as `nanopi_a64_defconfig`. Or you can send a pull request if you prefer.

sergey-suloev, 2021-10-01:
@smaeul  in this case I can use pine64_plus_defconfig, correct ?

smaeul, 2021-10-03:
I prefer each board to have its own config file, even if there end up being several identical files. That way if the config does need to change in the future, there is no worry about how it affects some other board.

smaeul, 2021-10-03:
Implemented in #194 (merged as a194bdb192281baa3347a32aa6e45b9082517b4f).

closed 2021-10-03 by smaeul