crust-firmware/crust #195: ass support for Teres I DIY laptop kit

jonassmedegaard, 2021-10-05:
## Type of issue

Feature Request

### Description

Would be great to have Crust include a defconfig file for the [Teres I](https://linux-sunxi.org/Olimex_Teres-A64) DIY laptop.

I own such laptop and can help test, but am not skilled enough to figure out on my own how the schematics of the laptop mainboard corresponds to Crust configuration.

I can compile both Crust and u-boot (I am the package maintainer of Crust in Debian).
I have so far tried use the Crust build for Pinebook but I fail to wakeup the system after it has fallen asleep.

smaeul, 2021-10-07:
The TERES-I has the same PMIC and voltage regulator allocation as practically every other A64 board, so the only configuration needed is selecting `CONFIG_MFD_AXP20X`. For reference, I'm comparing the schematic to the constants in [`regulator_list.c`](https://github.com/crust-firmware/crust/blob/master/common/regulator_list.c).

This is the same hardware configuration as the Pinebook, so the Pinebook firmware should work as-is. The Pinebook defconfig disables the serial console for additional power saving, but that is not necessary for basic functionality. In fact, enabling `CONFIG_SERIAL` and `CONFIG_DEBUG_LOG` would probably be helpful for the moment. Crust will share UART0 with Linux.

The most likely issue is with your kernel. What version of Linux are your running, and have you applied any patches? Try booting with the `clk_ignore_unused` kernel command line option, and see if that fixes things. If it does, great! You can:
* keep using that option,
* apply the first and fifth patches from [the `crust-minimal` branch here](https://github.com/crust-firmware/linux/commits/crust-minimal), or
* change [the 0 in the `apb0-twd` clock description](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/clk/sunxi-ng/ccu-sun8i-r.c?h=v5.14#n75) to a `CLK_IGNORE_UNUSED`.

If that doesn't fix your resume issues, please paste the serial console output from a suspend/resume attempt, ideally with debug logging enabled.

jonassmedegaard, 2021-10-09:
Thanks for clarifying that I need not fumble with serial port.

I recognize the frustration of getting patches approved - I've experienced it with a TERES-I USB keyboard fix for u-boot which took "forever" to get approved and required several rounds of rewrite (not pointing fingers, though: I am sure my limited skills was ultimately the cause - just saying I recognize the frustration of slow processing).

Perhaps sending a gently email to Stephen Boyd encouraging him to share his concerns for your patch series at the appropriate thread might help progress?

If a smaller patch is possible short term then that might be sensible - but since I don't understand the technical details here and don't know the reason the larger generic patchset hasn't been accepted I am obviously only guessing blindly here.

If not too much work to provide a workaround in Crust itself, then I would appreciate that, to be able to use Crust with mainline kernels, without needing to add custom patches.  Perhaps provide it behind a KConfig option default disabled, and its explanation pointing to the issue with the kernel, so as to raising the likelihood of your users getting informed about the underlying issue if they enable that workaround?

commit 485b40a5 (2022-06-12): configs: Add TERES-I defconfig

TERES-I is an A64-based laptop, which uses the usual PMIC and regulator
design. Suspend/resume has been verified to work with this defconfig,
which is the same as the Pinebook defconfig.

Closes: #195
closed 2022-06-12 by smaeul