crust-firmware/crust #202: [Pinephone] Phone wake-up caused by MicroSD card?

IntinteDAO, 2021-12-15:
## Bug Reports

Hello, I will try to report the bug as best as I can, if you have questions, please let me know.

I have noticed a regression in the 5.14 kernel version (and 5.15) in that Crust is not working properly [As far as I know, more people have this problem.]. Pinephone wakes up sometimes without much reason (of course in 5.13 it works fine). 

The only thing I noticed was the dmesg logs:
```
[47866.219177] PM: suspend entry (s2idle)
[47866.291163] Filesystems sync: 0.068 seconds
[47866.296004] Freezing user space processes ... (elapsed 0.028 seconds) done.
[47866.331385] OOM killer disabled.
[47866.332568] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[47866.334789] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[47866.350245] printk: Suspending console(s) (use no_console_suspend to debug)
[47866.857104] sunxi-mmc 1c0f000.mmc: data error, sending stop command
[47866.857142] mmc0: error -110 writing Power Off Notify bit
[47866.857158] PM: dpm_run_callback(): mmc_bus_suspend+0x0/0x74 returns -110
[47866.857192] mmcblk mmc0:aaaa: PM: failed to suspend async: error -110
[47866.873975] PM: Some devices failed to suspend, or early wake event detected
[47867.084750] OOM killer enabled.
[47867.087917] Restarting tasks ... done.
[47867.105376] PM: suspend exit
```

System: Mobian Bookworm (newest). I think I had the same thing on PostmarketOS (SXMO) after updating the kernel to 5.14, but there Crust didn't work at all.

smaeul, 2021-12-16:
This is a Linux kernel bug (assuming your microSD card is working properly), not a Crust bug. Probably the best place to file this issue is against your distro's kernel package. When you do this, please include the full `dmesg` log.

From the first line of your log, Linux was told to enter `s2idle`. `s2idle` is a shallow sleep state that does not use Crust at all. Linux is using `s2idle` either because your distro told it to (through e.g. systemd configuration) or because Crust is not installed/running. You can verify Crust is available by checking for the word `deep` in the file `/sys/power/mem_sleep`.

From the other lines in your log, there was some issue communicating with your microSD card, and Linux gave up entering the sleep state -- in other words: Linux did not wake up early; it never went to sleep in the first place.

IntinteDAO, 2021-12-16:
mobian@mobian:~$ cat /sys/power/mem_sleep
s2idle [deep]

In general should I report to the kernel provider or here? Or is it a s2idle problem?

smaeul, 2021-12-16:
In general, if you see any errors in `dmesg`, you should report the issue to your kernel provider. The usual symptom of a bug in Crust would be that the device hangs or reboots when attempting to wake it up.

In this specific case, the error occurs when powering down the MMC controller. That action happens as part of both s2idle and deep sleep, so the error is not specific to s2idle.

I'm closing this issue because it does not appear to be related to Crust.

closed 2021-12-16 by smaeul

smaeul, 2022-01-20:
In case you haven't seen it, this is the kernel patch that fixes the bug: https://lore.kernel.org/linux-mmc/20220115121447.641524-1-andrej.skvortzov@gmail.com/T/. This patch may already be applied by downstream maintainers.