crust-firmware/crust #204: Crust Wakeup issue

kmsgli, 2022-03-05:
## Type of issue
Question

gamelaster, 2022-03-05:
PinePhone Pro have different SoC which is not supported by Crust, so the issue on forums is not related to your problem.

kmsgli, 2022-03-05:
Well I will report that back to the guy who posted on the forums. Am I correct to assume my issue is related to crust as far as the standard pinephone?

smaeul, 2022-03-05:
Yes, it is possible that your issue is related to Crust.

What distro are you using? Some distros build Crust with [a few debugging options](https://github.com/crust-firmware/crust/blob/master/common/debug/Kconfig#L100-L120) already enabled. Otherwise, you will need to rebuild Crust with those options enabled. Then, after a failure to wake you will be able see what Crust was last doing and if it finished its part of the resume process with `hexdump /sys/devices/platform/soc/1f00000.rtc/nvmem0/nvmem`.

Please also provide a full `dmesg` taken after a successful resume.

With that information, I can suggest further troubleshooting steps.

kmsgli, 2022-03-05:
I am using manjaro phosh is that built in or do i need to rebuild crust?

smaeul, 2022-03-05:
It [does not have those options enabled](https://gitlab.manjaro.org/manjaro-arm/packages/core/uboot-pinephone/-/blob/master/PKGBUILD#L99). You can enable them by changing that block of lines and rebuilding the package:

```shell
  echo -e "\nBuilding CRUST for Pine64 PinePhone...\n"
  echo CONFIG_DEBUG_RECORD_STEPS >> configs/pinephone_defconfig
  echo CONFIG_DEBUG_VERIFY_DRAM >> configs/pinephone_defconfig
  make CROSS_COMPILE=or1k-elf- pinephone_defconfig
  make CROSS_COMPILE=or1k-elf- build/scp/scp.bin
  cp build/scp/scp.bin ../u-boot-${pkgver/rc/-rc}

```

kmsgli, 2022-03-05:
Ok just so I am clear because I have only built AUR package I would do this;
`git clone https://gitlab.manjaro.org/manjaro-arm/packages/core/uboot-pinephone.git`
 change the bit of code you listed above then
`makepkg -si`

test with 
`hexdump /sys/devices/platform/soc/1f00000.rtc/nvmem0/nvmem` 

then report back?

smaeul, 2022-03-05:
Yes, that should work. It looks like installing the updated package will try to flash the new U-Boot binary.

kmsgli, 2022-03-05:
> Yes, that should work. It looks like installing the updated package will try to flash the new U-Boot binary.

Is that an issue? or should it be fine to flash the new U-Boot binary?

Also I am reluctant to do this now as I am out of town and if something gets trashed I wont have a phone so I will wait until monday to do this and report back.

smaeul, 2022-03-05:
Crust is inside the U-Boot binary, so you have to flash U-Boot to get the updated Crust.

There's no rush; install the debug firmware and try to collect the logs whenever is convenient for you.

kmsgli, 2022-03-07:
I believe I have successfully installed U-Boot with the debug options. 

Running ` hexdump /sys/devices/platform/soc/1f00000.rtc/nvmem0/nvmem` after a wake from suspend I get:
```

0000000 0000 0000 0000 0000 0000 0000 0503 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
*
0000040
```

At this point I should wait for a freeze then reboot the phone and run` hexdump /sys/devices/platform/soc/1f00000.rtc/nvmem0/nvmem `again and report back?

smaeul, 2022-03-08:
Yes, that's the plan.

The NVMEM output you provided looks good (but see below). Examining the nonzero word, the first two digits confirm this is output from Crust v0.5. The last two digits come from [this list of steps](https://github.com/crust-firmware/crust/blob/master/include/common/steps.h#L16). 0x0503 means `STEP_SUSPEND_CSS`, so you caught Crust in the middle of turning off a CPU core. Because your kernel has cpuidle enabled, Linux tells Crust to power off the CPU cores 100+ times per second. This is normally good because it saves a bit of power, but it will unfortunately also overwrite the crash data before you have a chance to read it.

So you have two options:
1) If you have a UART adapter available, you can grab the data from the U-Boot prompt with `md.l 1f0010c 1`. Since you would be running this command before Linux has a chance to overwrite the data, you would get good data from the crash.
2) Otherwise, you will need to temporarily disable cpuidle by adding this kernel command line argument: `cpuidle.off=1`. You will also need to build a new version of Crust (sorry!) that contains commit e624ba0406f04071b2d4d4a8f1440cdca19d7507. (This commit keeps you from needing to disable other things to get good data.) You should be able to do this by setting `_scpver=master` in the PKGBUILD, and then fixing the URL to accept a branch instead of a tag:
```shell
"crust-${_scpver}.tar.gz::https://github.com/crust-firmware/crust/archive/refs/heads/${_scpver}.tar.gz"

```

kmsgli, 2022-03-08:
I don't have UART adapter as an option so number 2 is it. 
I am with you on rebuilding the software as its exactly what I did before by changing the initial echo lines I have already done and changing the two parameters at the very beginning of the PKGBUILD (that should be no problem).

I am a but fuzzy on turning `cpuilde.off=1` how to do I make that happen?

Interesting piece of information. Since rebuilding crust (yesterday at midnight) I have yet to get hung up in suspend. This could very well be a coincidence however its very rare that I have gone a full day without a hang up in suspend. I am going to try to stay off the charger as much as I can today to see if this is maybe more then coincidence and there is simply something wrong with the crust build that comes on the latest Manjaro betas ( seems unlikely as arch did the same thing and I have tried flashing OS fresh a few times).

kmsgli, 2022-03-10:
As of today I have not had a single hang up in suspend, could this problem have resolved itself with the recompile of crust? I tried fresh OS installs in the passed with no luck?

Does debug need to be on or was it simply manually rebuilding the package that fixed my issue?

smaeul, 2022-03-10:
> I am a but fuzzy on turning `cpuilde.off=1` how to do I make that happen?

From the Manjaro packaging, it looks like you would need to add the option to the `setenv bootargs` line in `/boot/boot.txt` and then run `pp-uboot-mkscr` as described in that file.

> Does debug need to be on or was it simply manually rebuilding the package that fixed my issue?

That's very interesting. Either one is possible. It's also possible (though unlikely at this point) that the issue still exists, but it has happened not to be triggered. You could try reinstalling the official package, or building a new package without the debug options, to see if the crashes come back. If they do, then likely the tiny additional delay introduced by the debugging code is allowing suspend/resume to complete.

kmsgli, 2022-03-11:
I just reinstalled using the git clone method without changing the debug options which in theory should be the same as the stock uboot setup when the OS is installed. I will report back in a day or two with if I have any suspend resume problems.

kmsgli, 2022-03-13:
With debug off I had it stall twice in the last few days I am going to recompile with debug on and see what happens.

kmsgli, 2022-03-15:
OK so it seems it was just dumb luck. With debug enabled I just had my first suspend hang so lets go back to plan A. As my pinephone is my daily driver I want you to look at this to make sure I am doing it correctly before I do it and botch my phone and loose my setup as I like the way the phone is setup right now.

First I will go nano int /boot/boot.txt and adjust the file to look like so;

```
#
# /boot/boot.txt
# After modifying, run "pp-uboot-mkscr" to re-generate the U-Boot boot script.
#

#
# This is the description of the GPIO lines used in this boot script:
#
# GPIO #98 is PD2, or A64 ball W19, which controls the vibrator motor
# GPIO #114 is PD18, or A64 ball AB13, which controls the red part of the multicolor LED
# GPIO #115 is PD19, or A64 ball AB12, which controls the green part of the multicolor LED
# GPIO #116 is PD20, or A64 ball AB11, which controls the blue part of the multicolor LED
#

gpio set 98
gpio set 114

# Set root partition to the second partition of boot device
part uuid ${devtype} ${devnum}:1 uuid_boot
part uuid ${devtype} ${devnum}:2 uuid_root

setenv bootargs loglevel=4 console=tty0 console=${console} earlycon=uart,mmio32,0x01c28000 consoleblank=0 boot=PARTUUID=${uuid_boot} root=PARTUUID=${uuid_root} rw rootwait quiet audit=0 bootsplash.bootfile=bootsplash-themes/manjaro/bootsplash cpuidle.off=1

if load ${devtype} ${devnum}:${distro_bootpart} ${kernel_addr_r} /Image; then
  gpio clear 98
  if load ${devtype} ${devnum}:${distro_bootpart} ${fdt_addr_r} /dtbs/${fdtfile}; then
    if load ${devtype} ${devnum}:${distro_bootpart} ${ramdisk_addr_r} /initramfs-linux.img; then
      gpio set 115
      booti ${kernel_addr_r} ${ramdisk_addr_r}:${filesize} ${fdt_addr_r};
    else
      gpio set 116
      booti ${kernel_addr_r} - ${fdt_addr_r};
    fi;
  fi;
fi

# EOF
```

Then I wll run;
`pp-uboot-mksrc`

Next I will follow the same process I did above for turning on debug mode as well as changing the _scpver=master and the url link leaving me with a PKGBUILD that looks like so:
```

# Maintainer: Philip Mueller <philm@manjaro.org>
# Contributor: Danct12 <danct12@disroot.org>
# Contributor: Furkan K. <furkan@fkardame.com>
# Contributor: Dragan Simic <dsimic@buserror.io>

pkgname=uboot-pinephone
pkgver=2022.04rc3
pkgrel=2
_tfaver=2.6
_scpver=master
_commit=589c659035a44a683b087fd75fe0b7667f7be7f5
pkgdesc="U-Boot for Pine64 PinePhone with CRUST support"
arch=('aarch64')
url="crust-${_scpver}.tar.gz::https://github.com/crust-firmware/crust/archive/refs/heads/${_scpver}.tar.gz"

license=('GPL')
depends=('uboot-tools')
makedepends=('bc' 'python' 'swig' 'dtc' 'arm-none-eabi-gcc' 'or1k-elf-gcc' 'or1k-elf-binutils' 'bison' 'flex' 'python-setuptools')
provides=('uboot')
conflicts=('uboot')
[... 124 lines cut ...]
  install -D -m 0644 u-boot-sunxi-with-spl-pinephone{-624,-592,-552,-528,-492}.bin -t "${pkgdir}/boot"
  install -D -m 0644 "${srcdir}/boot.txt" -t "${pkgdir}/boot"
  install -D -m 0755 "${srcdir}/pp-uboot-mkscr" -t "${pkgdir}/usr/bin"

  install -D -m 0755 "${srcdir}/pp-prepare-fstab" -t "${pkgdir}/usr/bin"
  install -D -m 0644 "${srcdir}/pp-prepare-fstab.service" -t "${pkgdir}/usr/lib/systemd/system"
  install -D -m 0755 "${srcdir}/pp-uboot-flash" -t "${pkgdir}/usr/bin"
  install -D -m 0644 "${srcdir}/uboot.conf" -t "${pkgdir}/etc/pinephone"
}

```

Finally I will run 
`makepkg -si`

and reboot.

Then on the next freeze I reboot and run 
`hexdump /sys/devices/platform/soc/1f00000.rtc/nvmem0/nvmem`

To get some use-able information to try and trouble shoot this?

Sorry for the long winded post I just dont want to botch the install as everything is just the way I want it.

Thanks so much for the help

kmsgli, 2022-03-16:
Ok so I decided to take a dd image of my eemc as a safety net and followed the above instructions I laid out for myself. Initially crust-master had an integrity check issue, so I ran makepkg with the --skipinteg switch to get it build (i know that's bad form but I really want to fix this issue).

here is the output of: 
` hexdump /sys/devices/platform/soc/1f00000.rtc/nvmem0/nvmem`

```
0000000 0000 0000 0000 0000 0000 0000 0503 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
*
0000040
```

This was run after completing everything, rebooting, and then allowing the phone to suspend then wake again.

Am I ready to debug now?

smaeul, 2022-03-16:
Yes, all of that looks good. The integrity check failing is expected, since you did change which version of the crust source code got downloaded.

The purpose of adding the `cpuidle.off=1` option is so that the contents of `/sys/devices/platform/soc/1f00000.rtc/nvmem0/nvmem` are all zeroes (the default) after a clean reboot but _before_ suspending for the first time. If you like, you can verify that is the case. But I think you have everything set up correctly.

kmsgli, 2022-03-16:
Here is the output after a reboot but before a suspend, I assume because before it was rebooted the suspend worked correctly so it is the same 

```
0000000 0000 0000 0000 0000 0000 0000 0503 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
*
0000040

```

kmsgli, 2022-03-22:
Explain the cpuidle.off=1 setting a bit more. It seems I am not having the suspend hang at least not yet but I am noticing phosh restarting from time to time under heavy use-age which must be related to the cpuidle setting?

If this ends up being a fix can I just leave it that way?

kmsgli, 2022-03-26:
OK, I finally had it get stuck in suspend, the blue indicator light was flashing which is new but I could not wake it from suspend so I had to hold power until off then power it back on. The output once booted back up was the same as above;

```
0000000 0000 0000 0000 0000 0000 0000 0503 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
*
0000040

```

Maybe I don't have something done correctly?

smaeul, 2022-03-28:
Can you paste the full output from `dmesg`, and also the contents of `/proc/cmdline`? That will help me to understand what you are seeing.

I'll also try to answer your questions.

> Explain the cpuidle.off=1 setting a bit more.

The purpose of `cpuidle.off=1` ([documentation](https://www.kernel.org/doc/html/latest/admin-guide/kernel-parameters.html)) is to disable the cpuidle subsystem.

The cpuidle subsystem ([documentation](https://www.kernel.org/doc/html/latest/driver-api/pm/cpuidle.html)) opportunistically powers off CPU cores when the kernel thinks they will be unused for some period of time. The purpose is to save power; for the A64 SoC in the PinePhone, it reduces power consumption by up to 30 mW. That's not a lot, but it does mean a few extra minutes of battery. Every time Linux decides to power off a CPU, after doing its internal bookkeeping, it calls TF-A, which then calls Crust, to actually flip the power switch. Then, when a hardware interrupt occurs, Crust turns the core back on, so Linux can use it again. This happens hundreds of times per second.

When Crust flips the power switch, it records that `0503` value in the RTC. So the point of disabling cpuidle is to prevent Crust from overwriting the data from the crash with `0503`.

> It seems I am not having the suspend hang at least not yet but I am noticing phosh restarting from time to time under heavy use-age which must be related to the cpuidle setting?

Enabling/disabling cpuidle should have no noticeable effect on any application running on the PinePhone. cpuidle doesn't change any behavior visible from userspace; it only affects performance/power consumption.

> If this ends up being a fix can I just leave it that way?

Yes, absolutely. However, if the Crust debug settings or `cpuidle.off=1` or the combination of the two fixes suspend, then that points to where there may be a bug in Crust. So it is still valuable information.

> OK, I finally had it get stuck in suspend, the blue indicator light was flashing which is new but I could not wake it from suspend so I had to hold power until off then power it back on.

A flashing LED suggests a kernel panic, not an issue with Crust. During the part of system suspend where Crust gets involved, the main CPU is completely powered off. So there is no software running that could flash the LED.

> Maybe I don't have something done correctly?

This new issue seems unrelated to your original suspend issue. Manjaro would be the best place to ask about debugging the kernel panic.

As for the `0503`, the output from `dmesg` and the contents of `/proc/cmdline` would confirm if cpuidle was successfully turned off.

kmsgli, 2022-03-28:
Thanks so much for the detailed response!

Here is the output of dmesg;

```
[  379.151758] Bluetooth: hci0: Out-of-order packet arrived (0 != 7)
[  379.161046] Bluetooth: hci0: Out-of-order packet arrived (1 != 7)
[  379.169054] Bluetooth: hci0: Out-of-order packet arrived (2 != 7)
[  382.320622] PM: suspend entry (deep)
[  382.536322] Filesystems sync: 0.215 seconds
[  382.537124] Freezing user space processes ... (elapsed 0.006 seconds) done.
[  382.543555] OOM killer disabled.
[  382.543562] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  382.545458] printk: Suspending console(s) (use no_console_suspend to debug)
[  382.553726] musb-sunxi 1c19000.usb: sunxi-musb does not have ULPI bus control register
[  382.846033] Disabling non-boot CPUs ...
[  382.860476] psci: CPU1 killed (polled 0 ms)
[  382.892820] psci: CPU2 killed (polled 0 ms)
[  382.923723] psci: CPU3 killed (polled 4 ms)
[  382.949137] Enabling non-boot CPUs ...
[  382.957410] Detected VIPT I-cache on CPU1
[  382.958791] arch_timer: CPU1: Trapping CNTVCT access
[  382.959268] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[  382.975782] CPU1 is up
[  382.978735] Detected VIPT I-cache on CPU2
[  382.979447] arch_timer: CPU2: Trapping CNTVCT access
[  382.979713] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[  382.991918] CPU2 is up
[  382.995196] Detected VIPT I-cache on CPU3
[  382.995825] arch_timer: CPU3: Trapping CNTVCT access
[  382.996109] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[  383.009595] CPU3 is up
[  383.021618] sunxi-rsb 1f03400.rsb: RSB running at 4000000 Hz
[  383.238684] musb-sunxi 1c19000.usb: sunxi-musb does not have ULPI bus control register
[  383.327614] OOM killer enabled.
[  383.328230] Restarting tasks ... done.
[  383.387673] PM: suspend exit
[  384.044530] Bluetooth: hci0: RTL: examining hci_ver=07 hci_rev=000b lmp_ver=07 lmp_subver=8703
[  384.049649] Bluetooth: hci0: RTL: chip_type status=0 type=5
[  384.055142] Bluetooth: hci0: RTL: rom_version status=0 version=1
[  384.055386] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723cs_xx_fw.bin
[  384.057050] Bluetooth: hci0: RTL: loading rtl_bt/rtl8723cs_xx_config.bin
[  384.083103] Bluetooth: hci0: RTL: cfg_sz 63, total sz 19427
[  384.490741] Bluetooth: hci0: RTL: fw version 0xaa5ca4dc
[  468.617790] PM: suspend entry (deep)
[  468.992982] Filesystems sync: 0.374 seconds
[  468.994935] Freezing user space processes ... (elapsed 0.011 seconds) done.
[  469.006942] OOM killer disabled.
[  469.006954] Freezing remaining freezable tasks ... (elapsed 0.002 seconds) done.
[  469.009411] printk: Suspending console(s) (use no_console_suspend to debug)
[  469.017724] musb-sunxi 1c19000.usb: sunxi-musb does not have ULPI bus control register
[  469.300408] Disabling non-boot CPUs ...
[  469.313308] psci: CPU1 killed (polled 4 ms)
[  469.354041] psci: CPU2 killed (polled 4 ms)
[  469.391591] psci: CPU3 killed (polled 0 ms)
[  469.421129] Enabling non-boot CPUs ...
[  469.429198] Detected VIPT I-cache on CPU1
[  469.430569] arch_timer: CPU1: Trapping CNTVCT access
[  469.431037] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[  469.447748] CPU1 is up
[  469.450508] Detected VIPT I-cache on CPU2
[  469.451011] arch_timer: CPU2: Trapping CNTVCT access
[  469.451100] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[  469.461862] CPU2 is up
[  469.465081] Detected VIPT I-cache on CPU3
[  469.465761] arch_timer: CPU3: Trapping CNTVCT access
[  469.465982] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[  469.478402] CPU3 is up
[  469.486518] sunxi-rsb 1f03400.rsb: RSB running at 4000000 Hz
[  469.732257] musb-sunxi 1c19000.usb: sunxi-musb does not have ULPI bus control register
[... 2168 lines cut ...]
[49989.261881] option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
[49989.262888] option 2-1:1.2: device disconnected
[49989.317732] option1 ttyUSB3: GSM modem (1-port) converter now disconnected from ttyUSB3
[49989.318151] option 2-1:1.3: device disconnected
[49989.336246] qmi_wwan 2-1:1.4 wwan0: unregister 'qmi_wwan' usb-1c1b000.usb-1, WWAN/QMI device
[49989.468218] option 2-1:1.0: GSM modem (1-port) converter detected
[49989.499981] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB0
[49989.501348] option 2-1:1.1: GSM modem (1-port) converter detected
[49989.504175] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB1
[49989.507132] option 2-1:1.2: GSM modem (1-port) converter detected
[49989.512367] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB2
[49989.521475] option 2-1:1.3: GSM modem (1-port) converter detected
[49989.609548] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB3
[49989.633023] qmi_wwan 2-1:1.4: cdc-wdm0: USB WDM device
[49989.641236] qmi_wwan 2-1:1.4 wwan0: register 'qmi_wwan' at usb-1c1b000.usb-1, WWAN/QMI device, 56:36:4f:ef:ad:ba
[51822.281606] anx7688 0-0028: BC 1.2 result: SDP
[52128.689366] anx7688 0-0028: BC 1.2 result: SDP
```

Here is the contents of /proc/cmdline;
`loglevel=4 console=tty0 console=ttyS0,115200 earlycon=uart,mmio32,0x01c28000 consoleblank=0 boot=PARTUUID=ed58b28e-01 root=PARTUUID=ed58b28e-02 rw rootwait quiet audit=0 bootsplash.bootfile=bootsplash-themes/manjaro/bootsplash cpuilde.off=1`

[... cut ...]  no useable crash data. Once  I recmompiled crust like described above with no more suspend issue except the one that sounds like was not a crust issue. The other interseting piece of information is I actually forgot a step when I recompiled so I did the master branch changes but forgot the debug changes and that sent me back into suspend lock out.

It seems debug is the key to this, So your original estimation of an extra bit of time to write to the log is preventing this issue from happeneing.

smaeul, 2022-03-29:
> `cpuilde.off=1`

The typo here explains why you were still seeing `0503` after a reboot.

> It seems debug is the key to this, So your original estimation of an extra bit of time to write to the log preventing this issue from happening is the most likely answer.
>
> Let me know what you think.

I will try to figure out where adding delay makes a difference. I don't know how long this will take.

kmsgli, 2022-03-29:
Well that is embarrassing. 

After re-running the /boot/boot.txt bit without the typo and restarting, then letting the phone suspend and wake I get this;

```
0000000 0000 0000 0000 0000 0000 0000 0519 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
*
0000040
```
 Seems my typo slowed down this progress. Sorry about that.

kmsgli, 2022-04-02:
Finally had a suspend lockup here was the output after reboot;

```
0000000 0000 0000 0000 0000 0000 0000 0519 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
*
0000040
```

Same as above? Maybe I still have something wrong?

smaeul, 2022-04-02:
You probably have everything correct. That code means Crust completed all of its resume steps and turned the ARM CPU back on. So Crust itself is not hanging. But if there is a bug in how Crust restores the hardware state, that could prevent Linux from running correctly after the resume.

I think you've given me all of the data that you can at this point, so thank you. Like I said, I will have to do some more digging in to what is going wrong.

kmsgli, 2022-04-03:
Thanks for all the help, it is certainly better with debug on as it only happens once or twice a week now as opposed to upwards of a few times a day.  Let me know if you need any other testing done.

XaviDCR92, 2022-04-10:
As mentioned [here](https://github.com/milky-sway/pinephone-scripts/pull/3#issuecomment-1093949710), I am having the same issue as described above, also a few times a day.

The phone crashes when entering suspend and thus is unable to wake up, even if the power button is pressed. Thanks to [this script](https://github.com/milky-sway/pinephone-scripts/blob/main/sleepwalk), typically the phone would sleep for a specific amount of time and then wake up, as shown below:

```
Apr 08 15:42:26 danctnix sleepwalk[2955]: Sleeping for 90s
Apr 08 15:42:26 danctnix kernel: PM: suspend entry (deep)
Apr 08 15:43:57 danctnix kernel: Filesystems sync: 0.054 seconds
Apr 08 15:43:57 danctnix kernel: Freezing user space processes ... (elapsed 0.018 seconds) done.
```

However, sometimes it cannot wake up and thus must be force-restarted. For example, as shown below, the phone crashed at `Apr 08 14:48:37` and did not report anything else on `journalctl` until rebooted.

```
Apr 08 14:48:37 danctnix sleepwalk[2977]: Sleeping for 90s
Apr 08 14:48:37 danctnix kernel: PM: suspend entry (deep)
-- Boot 8657034898c54b2d83e3c0b14e715531 --
Apr 08 15:37:09 danctnix kernel: Booting Linux on physical CPU 0x0000000000 [0x410fd034]
```

kmsgli, 2022-04-10:
Have you tried recompiling crust using the debug flag as I posted above? That has cut down the amount this happens to nearly nothing.

XaviDCR92, 2022-04-15:
I have tried to clone the repository and run `makepkg -si` on it, although `pacman` is unable to find the following packages:

```
error: target not found: or1k-elf-gcc
error: target not found: or1k-elf-binutils
```

I can see a [`x86_64` version](https://archlinux.org/packages/community/x86_64/or1k-elf-gcc/) of such packages, though. Am I then supposed to build crust on a `x86_64` computer? Or from a PinePhone?

kmsgli, 2022-04-15:
@XaviDCR92 
I think that issue is because you do not have the devel package installed, I could be wrong but try running this first;

`sudo pacman -S base-devel`

That gives you those tools that are required to run makepkg

kmsgli, 2022-04-17:
So this problem has come back up. I just took a large update and uboot was updated. I followed these instructions and rebuilt the package as with the update suspend is hanging frequently again. Is this because the github package has updated and I am not installing a different version?

How can I get back to how it was before the update?

Should i use downgrade to install 
uboot-pinephone-2022.04rc4-1

Thanks

XaviDCR92, 2022-04-18:
> @XaviDCR92 I think that issue is because you do not have the devel package installed, I could be wrong but try running this first;
> 
> `sudo pacman -S base-devel`
> 
> That gives you those tools that are required to run makepkg

`base-devel` was already installed on the system, so it should not be related to this.

```
$ makepkg -si                                                                         
==> Making package: uboot-pinephone 2022.04-1 (Mon 18 Apr 2022 10:54:51 AM CEST)
==> Checking runtime dependencies...         
==> Checking buildtime dependencies...                  
==> Installing missing dependencies...         
[sudo] password for alarm:
error: target not found: or1k-elf-gcc
error: target not found: or1k-elf-binutils
==> ERROR: 'pacman' failed to install missing dependencies.
==> Missing dependencies:
  -> swig
  -> dtc
  -> arm-none-eabi-gcc
  -> or1k-elf-gcc
  -> or1k-elf-binutils
==> ERROR: Could not resolve all dependencies.
```

As [I stated above](https://github.com/crust-firmware/crust/issues/204#issuecomment-1100432672), it looks like there are some missing packages. Whereas the packages below are in fact available:

```
$ pacman -Ss swig
extra/swig 4.0.2-5
    Generate scripting interfaces to C/C++ code
$ pacman -Ss dtc
community/dtc 1.6.1-4
    Device Tree Compiler
$ pacman -Ss arm-none-eabi-gcc
danctnix/arm-none-eabi-gcc 11.2.0-1
    The GNU Compiler Collection - cross compiler for ARM EABI (bare-metal) target
```

All of the commands below return nothing:

```
$ pacman -Ss or1k-elf-gcc
$ pacman -Ss or1k-elf-binutils
$ pacman -Ss or1k-elf
```

It looks like an `aarch64` version of the `or1k-elf-*` packages is not available.

kmsgli, 2022-04-18:
@XaviDCR92 my apologies I thought maybe you did not have the devel packages installed, I was mistaken.

Those packages you are missing are not installed on my system and my u-boot works so something else must be going on.

I just resolved my new suspend issues from updating by restoring to a old dd image of my system from just before i built u-boot from the PKGBUILD described in this thread.

uboot-pinephone 2022.04-1 which is what you are trying to install gave me massive issues. I can not tell if it was because of uboot-pinephone 2022.04-1 or because of other packages that updated around it.

my current version that was built but not installed just prior to taking my dd image when this thread started is uboot-pinephone 2022.04rc3-2, Installing that on the old dd iimage without updating my system has returned my phone to how it was which is nearly working perfect. Maybe try building that package and do not update the system if you have not already done sp.