crust-firmware/crust #206: Waydroid wakes up phone until I turn off screen

IntinteDAO, 2022-03-22:
I wonder where to report the error. Is it a problem with Gnome or with CRUST or maybe something else.

Waydroid is an environment that allows you to run Android on Linux.

Every now and then Waydroid wakes up the phone to download new viruses from Google or something ;) Unfortunately Pinephone (regular) doesn't turn off the screen or go into Deep Sleep after such a wakeup. It stays in the woken up state with the lock screen.

https://0bin.net/paste/lcOQ4xfh#dNKQtFFPwIhclEWBaZEYKOa1LJDeVLzW1wRDpPB8T7j

smaeul, 2022-03-23:
This is a problem with GNOME or systemd-logind/elogind. Those are the two programs that decide when to turn the screen on or off, and when to go to sleep. Crust is not involved in making those decisions. It only gets involved when Linux has already gone to sleep.

IntinteDAO, 2022-03-23:
Thanks

I'm closing the application and looking in Gnome.

closed 2022-03-23 by IntinteDAO