crust-firmware/crust #213: How to overcome firmware overflows error?

nenadperic, 2023-03-31:
## Type of issue

Question

I am trying to compile Crust firmware for OrangePi Prime. It seems to be fairly similar to OrangePi PC2, this is the defconfig I'm using.

I'm having issues (that are most likely because of my toolchain) with compiling Crust, but I haven't been able to resolve them myself. I've used the [Meta Firmware Build System](https://github.com/crust-firmware/meta) to build the package.

I've installed cross or1k-linux-musl 11.2.1 20211120 package obtained from https://musl.cc/#binaries

When I run `make V=1` this is the error I'm getting.

```
grep -Fqsx BOARD=orangepi_prime .config || printf 'BOARD=%s\n' orangepi_prime >.config
make -C arm-trusted-firmware CROSS_COMPILE=aarch64-linux-musl- \
	ARCH=aarch64 \
	BUILD_MESSAGE_TIMESTAMP='"0"' \
	DEBUG=0 \
	PLAT=sun50i_a64 \
	bl31
make[1]: Entering directory '/home/nperic/work/meta/arm-trusted-firmware'
mkdir -p  "/home/nperic/work/meta/arm-trusted-firmware/build/sun50i_a64/release/libfdt"
mkdir -p  "/home/nperic/work/meta/arm-trusted-firmware/build/sun50i_a64/release/lib"
mkdir -p  "/home/nperic/work/meta/arm-trusted-firmware/build/sun50i_a64/release/romlib"
mkdir -p  "/home/nperic/work/meta/arm-trusted-firmware/build/sun50i_a64/release/libc"
mkdir -p  "/home/nperic/work/meta/arm-trusted-firmware/build/sun50i_a64/release/libwrapper"
make[1]: Leaving directory '/home/nperic/work/meta/arm-trusted-firmware'
make -C crust CROSS_COMPILE=or1k-linux-musl- \
	HOST_COMPILE=aarch64-linux-musl- \
	build/scp/scp.bin
make[1]: Entering directory '/home/nperic/work/meta/crust'
exec scripts/version.sh /home/nperic/work/meta/crust build/include/version.h
exec or1k-linux-musl-gcc -Os -pipe -std=c11 -fdata-sections -ffunction-sections -fno-builtin -fno-common -fvar-tracking-assignments -ggdb -Wall -Wextra -Wformat=2 -Wpedantic -Wshadow -Werror=implicit-function-declaration -Werror=implicit-int -Werror=pointer-arith -Werror=pointer-sign -Werror=strict-prototypes -Werror=undef -Werror=vla -Wno-missing-field-initializers -ffreestanding -flto -fno-asynchronous-unwind-tables -fno-pie -fomit-frame-pointer -funsigned-char -Werror=implicit-fallthrough=5 -ffixed-r2 -msfimm -mshftimm -msoft-div -msoft-mul -nostdlib -no-pie -static -Wl,-O1 -Wl,--build-id=none -Wl,--fatal-warnings -Wl,--gc-sections -Wl,--no-dynamic-linker -Wl,--no-undefined  -Wl,-Map,build/scp/scp.map -o build/scp/scp.elf -T build/scp/arch/or1k/scp.ld.o build/scp/arch/or1k/counter.o build/scp/arch/or1k/exception.o build/scp/arch/or1k/math.o build/scp/arch/or1k/runtime.o build/scp/arch/or1k/start.o build/scp/common/debug.o build/scp/common/delay.o build/scp/common/device.o build/scp/common/regulator_list.o build/scp/common/scpi.o build/scp/common/scpi_cmds.o build/scp/common/simple_device.o build/scp/common/system.o build/scp/common/timeout.o build/scp/drivers/cir/cir.o build/scp/drivers/cir/rc6.o build/scp/drivers/cir/sunxi-cir.o build/scp/drivers/clock/clock.o build/scp/drivers/clock/ccu.o build/scp/drivers/clock/ccu_helpers.o build/scp/drivers/clock/r_ccu_common.o build/scp/drivers/clock/sun50i-a64-ccu.o build/scp/drivers/clock/sun8i-r-ccu.o build/scp/drivers/counter/sun6i-a31-cnt64.o build/scp/drivers/css/css.o build/scp/drivers/css/css_default.o build/scp/drivers/css/css_helpers.o build/scp/drivers/css/css_power_state.o build/scp/drivers/css/sun50i-a64-css.o build/scp/drivers/dram/dram.o build/scp/drivers/dram/sun8i-h3-dram.o build/scp/drivers/gpio/gpio.o build/scp/drivers/gpio/sunxi-gpio.o build/scp/drivers/irq/irq.o build/scp/drivers/irq/sun6i-a31-r-intc.o build/scp/drivers/msgbox/msgbox.o build/scp/drivers/msgbox/sunxi-msgbox.o build/scp/drivers/pmic/pmic.o build/scp/drivers/regmap/regmap.o build/scp/drivers/regmap/regmap-i2c.o build/scp/drivers/regmap/sun6i-i2c.o build/scp/drivers/regulator/regulator.o build/scp/drivers/regulator/gpio.o build/scp/drivers/regulator/sy8106a.o build/scp/drivers/serial/serial.o build/scp/drivers/serial/uart.o build/scp/drivers/serial/sun8i-h3-uart.o build/scp/drivers/watchdog/watchdog.o build/scp/drivers/watchdog/sun6i-a31-wdt.o build/scp/lib.a
/usr/bin/../lib/gcc/or1k-linux-musl/11.2.1/../../../../or1k-linux-musl/bin/ld: Firmware overflows allocated memory area
/usr/bin/../lib/gcc/or1k-linux-musl/11.2.1/../../../../or1k-linux-musl/bin/ld: Firmware overflows into SCPI shared memory
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:201: build/scp/scp.elf] Error 1
make[1]: Leaving directory '/home/nperic/work/meta/crust'
make: *** [Makefile:100: crust/build/scp/scp.bin] Error 2
```

smaeul, 2023-07-08:
The most likely problem is that the toolchain has broken/unsupported LTO (which would unfortunately mean I could not recommend the musl.cc toolchains anymore). Uploading `crust/.config` and `crust/build/scp/scp.map` would give me an idea about what is going wrong.

nenadperic, 2023-09-16:
Here is my `crust/.config`
```
#
# Automatically generated file; DO NOT EDIT.
# Crust Firmware Configuration
#
CONFIG_ARCH="or1k"
CONFIG_ARCH_OR1K=y
CONFIG_PLATFORM="a64"
# CONFIG_PLATFORM_A23 is not set
CONFIG_PLATFORM_A64=y
# CONFIG_PLATFORM_A83T is not set
# CONFIG_PLATFORM_H3 is not set
# CONFIG_PLATFORM_H6 is not set
# CONFIG_SOC_A64 is not set
CONFIG_SOC_H5=y
CONFIG_HAVE_DRAM_SUSPEND=y
CONFIG_HAVE_HDMI=y
CONFIG_HAVE_R_CIR=y
CONFIG_HAVE_R_I2C=y
CONFIG_HAVE_R_TWD=y

#
# Device drivers
#
# CONFIG_CEC is not set
CONFIG_CIR=y
# CONFIG_CIR_PROTO_NEC is not set
# CONFIG_CIR_PROTO_NECX is not set
CONFIG_CIR_PROTO_RC6=y
CONFIG_CIR_WAKE_CODE=0x800f040c
CONFIG_CIR_CLK_RATE=125000
CONFIG_CIR_USE_OSC24M=y
CONFIG_R_CIR_RX_PIN=11
CONFIG_OSC24M_SRC_X24M=y
# CONFIG_I2C_PINS_NONE is not set
CONFIG_I2C_PINS_PL0_PL1=y
CONFIG_I2C_PIN_MODE_PL0_PL1=2
CONFIG_I2C=y
# CONFIG_MFD_AXP20X is not set

#
# Power management controllers
#
# end of Power management controllers

#
# Regulators (voltage/current)
#
CONFIG_REGULATOR_GPIO=y
# CONFIG_REGULATOR_GPIO_CPU is not set
CONFIG_REGULATOR_GPIO_DRAM=y
CONFIG_REGULATOR_GPIO_DRAM_PIN=9
# CONFIG_REGULATOR_GPIO_VCC_PLL is not set
# CONFIG_REGULATOR_GPIO_VDD_SYS is not set
CONFIG_REGULATOR_SY8106A=y
# end of Regulators (voltage/current)

CONFIG_SERIAL=y
CONFIG_SERIAL_DEV_UART0=y
# CONFIG_SERIAL_DEV_UART1 is not set
# CONFIG_SERIAL_DEV_UART2 is not set
# CONFIG_SERIAL_DEV_UART3 is not set
# CONFIG_SERIAL_DEV_R_UART is not set
CONFIG_SERIAL_BAUD=0
CONFIG_WATCHDOG_SUN6I_A31_WDT=y
# CONFIG_WATCHDOG_SUN9I_A80_TWD is not set
# end of Device drivers

#
[... 3355 lines cut ...]
 .rodata.str1.1
                0x0000000000013aca        0x9 build/scp/drivers/regulator/gpio.o
 .rodata.str1.1
                0x0000000000013ad3        0x8 build/scp/drivers/regulator/sy8106a.o
 .rodata.str1.1
                0x0000000000013adb        0x6 build/scp/drivers/serial/sun8i-h3-uart.o
 .rodata.str1.1
                0x0000000000013ae1        0x7 build/scp/drivers/watchdog/sun6i-a31-wdt.o
 .rodata.system_state_machine.str1.1
                0x0000000000013ae8        0xe build/scp/common/system.o
                0x0000000000013af8                . = ALIGN (0x4)
 *fill*         0x0000000000013af6        0x2 

.data           0x0000000000013af8        0xc
 *(SORT_BY_ALIGNMENT(SORT_BY_NAME(.data*)))
 .data.iosc_rate
                0x0000000000013af8        0x4 build/scp/drivers/clock/r_ccu_common.o
 .data.power_state
                0x0000000000013afc        0x6 build/scp/drivers/css/css_power_state.o
                0x0000000000013afc                power_state
 .data.system_state
                0x0000000000013b02        0x1 build/scp/common/system.o
                0x0000000000013b04                . = ALIGN (0x4)
 *fill*         0x0000000000013b03        0x1 
                0x0000000000000001                ASSERT ((SIZEOF (.data) == ALIGN (0xc,

smaeul, 2023-09-16:
Yes, that's definitely broken LTO. All of the `.gnu.lto_` sections are unused, and there is no cross-file inlining. For comparison, here's an extract from my `scp.map` file:
```
.text           0x00010000     0x2b84
 *(.text.start)
 .text.start    0x00010000       0x70 build/scp/arch/or1k/start.o
                0x00010000                start
 *(SORT_BY_ALIGNMENT(SORT_BY_NAME(.text*)))
 .text.__udivsi3
                0x00010070       0x6c build/scp/arch/or1k/runtime.o
                0x00010070                __udivsi3
 .text.bitmap_clear
                0x000100dc       0x30 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.bitmap_set
                0x0001010c       0x2c /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_apb2
                0x00010138        0xc /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_apb2_parent
                0x00010144        0xc /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_dram_parent
                0x00010150       0x28 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_null_parent
                0x00010178        0x8 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_parent
                0x00010180       0x24 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_parent_rate
                0x000101a4        0x8 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_pll_periph0_rate
                0x000101ac        0xc /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_r_cir_parent
                0x000101b8       0x28 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_r_cir_rate
                0x000101e0       0x48 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_rate
                0x00010228       0x28 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_get_state
                0x00010250       0x98 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.ccu_set_state
                0x000102e8      0x11c /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.clock_enable
                0x00010404       0x78 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.clock_get
                0x0001047c       0x98 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.clock_get_state
                0x00010514       0x70 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.clock_put
                0x00010584       0xa8 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.css_set_power_state
                0x0001062c      0x414 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.device_get
                0x00010a40       0x94 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.device_get_or_null
                0x00010ad4       0x34 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.device_put
                0x00010b08       0x40 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.dummy_probe
                0x00010b48        0x8 /tmp/cchMcIeE.ltrans0.ltrans.o
 .text.dummy_release
                0x00010b50        0x8 /tmp/cchMcIeE.ltrans0.ltrans.o
```
You can see all of the functions come from one file (`/tmp/cchMcIeE.ltrans0.ltrans.o`) which was generated by the LTO process, and many functions are missing because they were inlined. For example, the whole `drivers/css` directory got inlined into `css_set_power_state`.

nenadperic, 2023-09-16:
Thanks for checking, I'll close this issue then (nothing to do).

closed 2023-09-16 by nenadperic

smaeul, 2023-09-16:
I hacked up the build system to use `-fwhole-program` instead of `-flto`, and the result is almost identical to a toolchain with functioning LTO (in fact, it's slightly worse). So I plan to stick with using LTO for now.

```patch
diff --git a/Makefile b/Makefile
index 5197de4..870819c 100644
--- a/Makefile
+++ b/Makefile
@@ -65,13 +65,14 @@ AFLAGS               = -Wa,--fatal-warnings \
                   -Wa,--noexecstack
 CFLAGS          = $(COMMON_CFLAGS) \
                   -ffreestanding \
-                  -flto \
+                  -fwhole-program \
                   -fno-asynchronous-unwind-tables \
                   -fno-pie \
                   -fomit-frame-pointer \
                   -funsigned-char \
                   -Werror=implicit-fallthrough=5
 CPPFLAGS        = $(COMMON_CPPFLAGS) \
+                  -I$(SRC) \
                   -I$(SRC)/include/drivers \
                   -I$(SRC)/include/stdlib \
                   $(foreach header,$(HEADERS),-include $(notdir $(header))) \
@@ -199,15 +200,16 @@ $(TGT)/%.bin: $(TGT)/%.elf
 
 $(TGT)/%.elf $(TGT)/%.map: $(obj-all) $(TGT)/lib.a
        $(M) LD $@
-       $(Q) $(CC) $(CFLAGS) $(LDFLAGS) -Wl,-Map,$(TGT)/$*.map -o $@ -T $^
+       $(Q) printf '#include "%s"\n' > $@.c $(wildcard $(patsubst $(TGT)/%.o,$(SRC)/%.c,$^))
+       $(Q) $(CC) $(CFLAGS) $(AFLAGS) $(CPPFLAGS) $(LDFLAGS) -Wl,-Map,$(TGT)/$*.map -o $@ -T $< $(wildcard $(patsubst $(TGT)/%.o,$(SRC)/%.S,$(filter-out %.ld.o,$^))) $@.c
 
 $(TGT)/lib.a:
        $(M) AR $@
-       $(Q) $(AR) Drcs $@ $^
+       $(Q) cat > $@ $^
 
 $(TGT)/%.o: $(SRC)/%.c
        $(M) CC $@
-       $(Q) $(CC) $(CPPFLAGS) $(CFLAGS) $(AFLAGS) -MMD -c -o $@ $<
+       $(Q) printf '#include "%s"\n' > $@ $<
 
 $(TGT)/%.o: $(SRC)/%.S
        $(M) AS $@
diff --git a/common/system.c b/common/system.c
index c1a633b..9916ec0 100644
--- a/common/system.c
+++ b/common/system.c
@@ -79,7 +79,7 @@ select_suspend_depth(uint8_t current_state)
        return SD_VDD_SYS;
 }
 
-noreturn void
+noreturn void __attribute__((externally_visible))
 system_state_machine(uint32_t exception)
 {
        const struct device *cec, *cir, *mailbox, *pmic, *watchdog;
diff --git a/drivers/css/css_default.c b/drivers/css/css_default.c
index 5c072db..554a2ce 100644
--- a/drivers/css/css_default.c
+++ b/drivers/css/css_default.c
@@ -23,6 +23,7 @@ css_get_core_count(uint32_t cluster UNUSED)
        return MAX_CORES_PER_CLUSTER;
 }
 
+#if 0
 /*
  * Generic implementation used when no platform support is available.
  */
@@ -71,3 +72,4 @@ void WEAK
 css_init(void)
 {
 }
+#endif
diff --git a/drivers/dram/dram.c b/drivers/dram/dram.c
index b68864f..c1ba55e 100644
--- a/drivers/dram/dram.c
+++ b/drivers/dram/dram.c
@@ -5,6 +5,7 @@
 
 #include <dram.h>
 
+#if 0
 void WEAK
 dram_suspend(void)
 {
@@ -19,3 +20,4 @@ void WEAK
 dram_init(void)
 {
 }
+#endif
diff --git a/drivers/irq/irq.c b/drivers/irq/irq.c
index a99b047..e534961 100644
--- a/drivers/irq/irq.c
+++ b/drivers/irq/irq.c
@@ -6,6 +6,7 @@
 #include <irq.h>
 #include <stdint.h>
 
+#if 0
 uint32_t WEAK
 irq_needs_avcc(void)
 {
@@ -23,3 +24,4 @@ irq_poll(void)
 {
        return 0;
 }
+#endif
```