crust-firmware/crust #217: Allwinner D1

sarahspock, 2023-08-13:
## Feature Request

Allwinner D1 Support

I would like advanced power management on Risc-v Allwinner D1 so that I can use Linux phone with it . This s what ships with the clockwork pi upmc

smaeul, 2023-09-01:
Allwinner D1 already has CPU suspend support [implemented in OpenSBI](https://github.com/riscv-software-src/opensbi/blob/master/platform/generic/allwinner/sun20i-d1.c). Some additional firmware might still be required for DRAM suspend, since that code must run out of SRAM, but it may be possible to integrate that with OpenSBI as well. So crust may not be needed on D1.

If crust does need to be ported, then it would implement RPMI (RISC-V's SCMI copycat) instead of SCPI. And it would need to run on the HiFi4 DSP, so it would need to be ported to the Xtensa architecture first.

sarahspock, 2023-09-02:
Hi thank you for responding. Do you know if the OpenSBI comes with most distros or alternatively is a firmware that ships with the kernel (sorry not super knowledgable here)? I noticed that the device has pretty decent battery life already (better than pinephone?). I dont think DRAM suspend works though

smaeul, 2023-09-02:
Yes, OpenSBI is part of the system firmware, and [gets shipped with U-Boot](https://linux-sunxi.org/Allwinner_Nezha#U-Boot), so it is used by every distro.